cnlohr/channel3 #8: Flashing image

thatrez, 2016-10-22:
I used the Firmware Programmer to flash the two elf image and put the offset in the other one. I don't see any wifi hotspot broadcasting and tuning to channel 3 does nothing. Have I flashed this wrong?

thatrez, 2016-10-22:
I thought maybe it was because I initially flashed under windows. reflashed under linux using esptool, same exact issue

thatrez, 2016-10-22:
the TV demo program transmits - but no web interface

cnlohr, 2016-11-06:
Ah! Sorry, I was really busy for a long time.  I am back now.  You will have to burn the webpage onto the part with the  "burnweb" command.  You can also connect and cd to the web folder and type make and it will flash the webpage over the network.