cnlohr/channel3 #9: How to debug 0hz

mhoffma, 2017-04-25:
Once the system boots and connects and loads web page up I can see all the nice widgets in my browser. however my Hz fluctuations go from like 6 to 0 and once I get 0 it loses connection.

I'm using a nodemcu module USB powered, nothing connected.  Is the intended target esp01?

Any hints appreciated

cnlohr, 2017-04-29:
It should work fine on the NodeMCU the only concern may be the NTSC signal itself may be interfering with the wifi.  Please trying to run cnping against it and see if you get a see-saw effect.

mhoffma, 2017-04-29:
I got it working my setup was not flashed right on the ESP01 module, your
work is absolutely amazing.

cnlohr, 2017-04-29:
Can you be more specific about what you did?  I know this is a common problem for people.

mhoffma, 2017-04-29:
Yes. I moved to node mcu and added -fm dio to the esptool writeflash
command applying it by hand. Then I just did make burnweb.

I plan to try it again on esp01. If you want I can add my notes to your
project wiki.

cnlohr, 2017-04-29:
Using -fm dio where exactly?  It should either /just work/ or just /not work/

mhoffma, 2017-04-29:
It was crashing for me the inf crash loop

cnlohr, 2017-04-30:
But, it was getting far enough to load the webpage?

mhoffma, 2017-04-30:
yes I had a similar problem with your colorchord where it was unstable but
that turned out I believe to be a short on my board.  The performance Hz
was all over the place and then it would become totally unresponsive
dropping connection. I wired TX so I could see the terminal text and I
noticed the reboot sequence happening.  I'm not sure if the two things are
the same issue but my setup for channel3 has no wires just the breakout
board so its not my wiring as an issue.

It really is odd that web page loads and displays and you could even thumb
through the different pages on occasions and then it would die, it was
pretty frustrating.  I was just about to put trace into the code to figure
out where it was dying when I reflashed it. woops I just realized I did
change the code.

I don't see how this could be related but I made this change too.

--- a/user/user_main.c
:
+++ b/user/user_main.c
@@ -363,15 +363,16 @@ void ICACHE_FLASH_ATTR user_init(void)

        uart0_sendStr("\r\nesp8266 ws2812 driver\r\n");

-//     int opm = wifi_get_opmode();
+       int opm = wifi_get_opmode();
 //     if( opm == 1 ) need_to_switch_opmode = 120;
 //     wifi_set_opmode_current(2);
 //Uncomment this to force a system restore.
 //     system_restore();

+#define FORCE_SSID 1
 #ifdef FORCE_SSID
-#define SSID ""
-#define PSWD ""
+#define SSID "hoff"
+#define PSWD "xxxxxxx"
 #endif

        //Override wifi.
diff --git a/web/page.mpfs b/web/page.mpfs
index d4015cc..8cef52b 100644
Binary files a/web/page.mpfs and b/web/page.mpfs differ
:

cnlohr, 2017-04-30:
I don't see that either.  I guess we'll just have to chalk it up to an unknown cause/effect :(.  If you want we can close the issue

mhoffma, 2017-04-30:
yes

cnlohr, 2018-11-21:
I have noticed that what can happen is if your antenna is too short, or (unknown other details) the NTSC signal itself seems to interfere with the carrier detect for the 2.4 GHz wifi and prevents the ESP from sending any packets causing comms to fail.  I strongly recommend trying a longer wire length.

kuzmanzs, 2020-09-07:
Hi,
I have same issue. From 10 tries I got once connection and after page loading takes forever. And if I am lucky the refresh rate descending after few seconds and goes to offline. Video output working without any interruption.

Whitch wireleght do you recommend?  At the moment I am using 1pcs dupoint cable.

cnlohr, 2020-09-15:
I do not have any specific recommended length.  You can also try place something metallic near the antenna to mute it.  This is way more janky than you may have previously expected :-p