cnlohr/channel3 #13: Installed the bin files, but unable to connect to the web control

marek26340, 2018-02-16:
Hi, 

I want to try this out. It's so interesting that this tiny device can do even more things than your smartphone can't do. (well, only transmitting analog TV signal is a feature of the ESP, that smartphones can't do...)

Anyways, here's the problem.
Downloaded and installed your bin files to the correct addresses, reset it, and it started transmitting WiFi and the TV signal which contains the demo. I connect to ESP's WiFi and try to load http://192.168.4.1/, but it will just be loading forever or show an error "File not found".
I already tried erasing the whole ESP with a 1MB blank.bin to addresses 0x000000, 0x100000, 0x200000 and 0x300000, but after reflashing your bin files, the error is still there.

About the board: DOIT ESP8266 ESP-12E NodeMCU Devkit
Browser: Google Chrome for Android
Phone: Huawei P9 Lite running Android 7.0 (C432B392)

cnlohr, 2018-02-16:
Where did you flash the mpfs data?  You need three components to operate it.  0x000... bin, 0xX000 bin and the mpfs dat file.

Noonlord, 2018-04-12:
Where should I flash mpfs file?

cnlohr, 2018-04-13:
0x100000 or 0x10000 either will do.

marek26340, 2018-04-13:
I did not flash the mpfs data... omg