cnlohr/channel3 #14: I live in a PAL country. Could this work with PAL?

bootrino, 2018-03-04:


cnlohr, 2018-03-04:
you would need to change a number of constants to change the lines etc.  And you'd need to change the code to synth the signal but neither are too hard.  Check out the video to get a feel for how to use the web interface and you can do a number of tests that way.

Duality4Y, 2018-03-16:
so i have this portable tv but it's pall and i am trying to find where to change the lines, any pointers? i am looking in ntsc_broadcast but i am not really finding anything related to that.

Duality4Y, 2018-03-16:
ok this is way to complicated for me :D i adjusted the number of lines for pal, it's sharper now. so i'll have to live with it being in black and white for now :)

cnlohr, 2018-03-19:
I am shocked your TV was willing to tolerate that at all!

But, the other stuff you will have to tweak is in the table make for the frequency synthesizer -- it's one of the tools that runs that makes a table for the ESP.

PR #39 (2023-03-01): Added PAL support
Made some changes to the video timings to allow for PAL broadcasts. I didn't change the actual code for colour generation since I only wanted to use it with a B/W television.

Semi-Fixes #14
closed 2023-03-01 by cnlohr