cnlohr/channel3 #15: What is it that defines the resolution? Could TV audio be sent too?

bootrino, 2018-03-04:
I'm interested to know what it is that defines the screen resolution for this project?  is it adjustable?  Could it be higher somehow?

Is it possible also to send TV audio?

Coolest project ever! thanks!

cnlohr, 2018-03-04:
The resolution is subject to the standard for broadcast you use, i.e. NTSC is 525 (minus blanking lines leaving you with many fewer) and PAL is some specific number.

It is sadly not possible to send audio because there's just no bandwidth left to try to encode the FM, and it's much harder to encode than the video signal.

bootrino, 2018-03-13:
I hope you don't mind me asking these noobish questions.

I notice that the ESP32 has two I2S output ports.  I was wondering what might be made of this.... would it make any sense to use both ports to transmit the various stuff needed i.e. luma, chroma and audio.

Would anything be gained from using both IS2 ports on the ESP32? Is it even possible to generate one TV picture from two ports or does all the output need to come from a single port for some reason?  Or would using two ports result in some sort of timing problem in generation of the signal?

Also, on the readme for this project you say:
>> This is basically a 1-bit dithering DAC, operating at a frequency below the nyquist, trying to encode luma and color at the same time. Don't be surprised that the quality's terrible.

I'm trying to understand.... what is it that would need to be different to clean up the signal so that it had higher quality and not have patterns crawling down the screen?  If you could magically change something about your broadcast TV project to fix this, what would it be - more processing power? Something else?  I'm just curious to know what would need to change to stabilise the colors in your broadcast TV image.

cnlohr, 2018-03-13:
yes, absolutely, it would!  Also, you could make a significant improvement by using a single I2C engine and using 4- or 8- of the output bits to produce a nice clean signal.

The only problem then is it is still a little wonky because you're creating signals above nyquist, so I don't really know what kinds of tricks you would have to use to keep the signal clean.

It would take a lot of experimentation with a SDR.  Maybe you could use the clock pin to chop the output to force the signal to flip around the nyquist.  Right now, it's a total abuse that the output pins on the ESP8266 are TOTALLY slew-unlimited, so they make these ridiculous square waves that shoot noise all over the spectrum, so transmitting above nyquist isn't that bad.

I can't remember his name but someone made a very high quality AM transmitter with the ESP32 by using the I2S engine and an ADC so they could make a clean signal.

The reason for Channel 3 is because the math works out _perfectly_ to use a 80 MHz clock, outputting at a 62.5 MHz baseband and precisely an (315/88) MHz NTSC carrier if you use a cycle time of 1408.

I am not sure if it was purely coincidental, but it was amazing when I found out.  You could use other frequencies but either the quality wouldn't be as good or the tuning would need to be slightly off.

bootrino, 2018-03-13:
Thanks Charles for the interesting info - you might consider adding some of this thread to your front page readme - it's most informative.

>> using a single I2C engine and using 4- or 8- of the output bits to produce a nice clean signal.

Can you explain a little more about what you mean by this please?

>> someone made a very high quality AM transmitter
Perhaps this? https://github.com/bitluni/ESP32AMRadioTransmitter

Maybe with the 2 I2s ports on the esp32 then it would be possible to broadcast tv graphics signal with your Channel 3 solution, put an AM radio on top of the tv and transmit audio to the AM radio and have both audio and video for games or maybe video playback.

cnlohr, 2018-03-13:
I don't know how useful that would be, but I did want to fill you in on an experiment I did with an ESP8266 and nosdk8266.  I was able to transmit a baseband carrier and an FM signal and it got through loud and clear when I clocked the ESP and its AHB at 173 MHz.

bootrino, 2018-03-13:
>> I did want to fill you in on an experiment I did

Interesting... is this a recent experiment? Do you have any practical demos in mind for this?

>> I was able to transmit a baseband carrier and an FM signal and it got through loud and clear when I clocked the ESP and its AHB at 173 MHz

What is the significance of 173MHz - are you saying you've found a way to transmit FM along with a video image over broadcast?  I searched for 173MHz but couldn't find any particular reason why you chose this frequency.

>>when I clocked the ESP and its AHB at 173 MHz

Is it correct for this experiment that the chip must be clocked to match the desired FM frequency? If yes then presumably that means it is not possible to also transmit broadcast TV too?

>> with an ESP8266 and nosdk8266

What is the reason it was necessary to use the nosdk8266?  Is that because you wanted to switch off the wifi for some reason? Is transmitting wifi mutually exclusive with transmitting FM in this way?

cnlohr, 2018-03-13:
The experiment was lat week.  It's in the "try_at_audio" branch.

173/346 MHz is the fastest you can reliably clock the ESP. It just gives you more bits to turn on and off to help make a cleaner and cleaner signal where you want it.   Unfortunately 173 (or more precisely 520/3 MHz) does not correspond well to the 315/88 MHz that NTSC wants, so it would need to be fudged a little bit.

This branch does not produce an image, but does produce a carrier that the TV locks onto and the FM encoding.  It shouldn't be too hard to encode actual luma and chroma data as well.  The FM works surprisingly well even when very little of the signal is actually created. 

The reason for the nosdk8266 is because you cannot use nonstandard clock rates with the normal SDK.  The max you can clock the I2S bus with the normal SDK is 80 MHz, significantly slower than the crazy 173 you can get out of it when you let go of the normal SDK.

bootrino, 2018-03-15:
Hi there - I've been thinking and researching on something but not really able to work out the answer - you're the person most likely to know.

Is it possible to generate a display from a chip like an ESP32 on a digital tv frequency, perhaps using the wifi transmitter?

cnlohr, 2018-03-16:
I think about that problem frequently.  The problem is that you need to transmit 8VSB, which is non-trivial to transmit as you have to encode basically a 6 MHz bandwidth signal in a very precise way.   If you had an external oscillator used to bring it from base band up to transmit frequency, probably.  But then I don't even know if it would work because you might have to suppress the lower side band.

bootrino, 2018-03-24:
Hey Charles you probably follow bitluni already but he's doing some interesting stuff generating video and now sound from esp32 using the second DAC

https://youtu.be/MOoKQnWF4OY

cnlohr, 2018-03-25:
I wish this was higher on my list of things to do. I  really do <3 NTSC and other video formats.  But, no time ATM :(