cnlohr/channel3 #23: Can I have some instructions how to flash it

GenryEden, 2019-10-20:


cnlohr, 2019-11-12:
I do not have the exact command, and urge you or anyone else to post it, but you should be able to flash this entirely on using esptool.py.

mydevbox, 2020-01-26:
This is the command we used to flash to nodemcu on macos catalina using the .bin files from this repo.

our prerequisites:

1) Install the appropriate driver for nodemcu for usb to uart
2) Install esptool.py via homebrew

`sudo esptool.py --port /dev/tty.SLAB_USBtoUART write_flash 0x00000 image.elf-0x00000.bin 0x40000 image.elf-0x40000.bin`

We are able to attach to esp via wifi but going to 192.168.4.1 only returns this error. 

`File not found.`

Even though web interface isn't working we confirmed that our tv received the analog signal.

cnlohr, 2020-05-04:
You do need to also flash the web file.  You may need to do that via the network.  I think it's 
```make netweb```

SyedSajjadHaider, 2021-08-13:
@cnlohr i just bought esp8266 and saw your youtube video , you did amazing job .
My question i have cloned git repository , it has **image.elf-0x00000.bin** but it dont have **image.elf-0x40000.bin** can you give steps how you generated  **image.elf-0x40000.bin**  file.
again thanks a lot for great work :)