cnlohr/channel3 #24: Transmitting in the blanking interval for broadcasting teletext

sijanec, 2020-03-28:
Hello!
I am very amazed by your project and I've been wondering if it would be possible to transmit in the blanking interval and thus transmitting teletext? As per https://en.wikipedia.org/wiki/Teletext#Data_transmission, I see that it uses 296 bits per line for data transmission. Would it be possible to output 296 bits of data per line? I see you mention in the video that the resolution can get up to 232x220. Would that mean that there is not enough "pixels" to represent each bit? What about the lines? Since NTSC has 525 lines maximum and channel3 esp8266 broadcasts way less lines, would teletext even be achievable?

cnlohr, 2020-05-03:
It absolutely would be possible!  The problem is I didn't have any TVs that supported teletext or closed captioning.  You would need to modify the NTSC generation code, but it wouldn't be too hard to swap out the "do nothing" for "do something interesting"

sijanec, 2020-05-03:
Thank you very much for the answer!

luminxa, 2020-07-11:
did you ever get teletext working? i would be interested in this..

sijanec, 2020-07-11:
I realised I don't have teletext on my CRT TV so no, unfortunately I did
not try to do it.

hasaranga, 2020-07-28:
teletext require PAL right?

sijanec, 2020-07-29:
although the implementation differs from analogue video format, ntsc
versions have also been created.

see https://en.m.wikipedia.org/wiki/Teletext

cnlohr, 2020-08-14:
I don't have any way to receive it so I can't write it. Sorry.

cnlohr, 2020-08-14:
I'd appreciate it if someone else did it though