cnlohr/channel3 #26: build issue

kuzmanzs, 2020-09-02:
Hi,

I can't build it:
`common/http.c: In function 'http_disconnetcb':
common/http.c:328:37: error: 'struct espconn' has no member named 'reverse'
  ((struct HTTPConnection * )pespconn->reverse)->state = 0;
                                     ^
common/http.c: In function 'http_recvcb':
common/http.c:341:46: error: 'struct espconn' has no member named 'reverse'
  curhttp = (struct HTTPConnection * )pespconn->reverse;
                                              ^
common/http.c: In function 'httpserver_connectcb':
common/http.c:359:12: error: 'struct espconn' has no member named 'reverse'
    pespconn->reverse = &HTTPConnections[i];
            ^
Makefile:70: recipe for target 'image.elf' failed
make: *** [image.elf] Error 1
`
system config on Ubuntu 18
`myuser@myuser-NANUC8:~/dev/projects/c/esp/esp8266/projects/analogtv/channel3$ make all
echo $PATH
/home/myuser/dev/projects/c/esp/esp8266/esp-open-sdk/xtensa-lx106-elf/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/usr/lib/jvm/java-8-oracle/bin:/usr/lib/jvm/java-8-oracle/db/bin:/usr/lib/jvm/java-8-oracle/jre/bin
/home/myuser/dev/projects/c/esp/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -mlongcalls -I/home/c/dev/projects/c/esp/esp8266/projects/espterm-firmware/esp_iot_sdk_v1.5.2/include -Imyclib -Iinclude -Iuser -Os -I/home/myuser/dev/projects/c/esp/esp8266/projects/espterm-firmware/esp_iot_sdk_v1.5.2/include/ -Icommon -DICACHE_FLASH driver/uart.c common/http.c common/mystuff.c common/flash_rewriter.c common/commonservices.c common/http_custom.c common/mdns.c common/mfs.c user/custom_commands.c user/ntsc_broadcast.c user/user_main.c user/3d.c tablemaker/broadcast_tables.c tablemaker/CbTable.c  -nostdlib -L/home/myuser/dev/projects/c/esp/esp8266/projects/espterm-firmware/esp_iot_sdk_v1.5.2/lib -L/home/myuser/dev/projects/c/esp/esp8266/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/libgcc.a -g /home/myuser/dev/projects/c/esp/esp8266/projects/espterm-firmware/esp_iot_sdk_v1.5.2/lib/libmain.a /home/myuser/dev/projects/c/esp/esp8266/projects/espterm-firmware/esp_iot_sdk_v1.5.2/lib/libpp.a /home/myuser/dev/projects/c/esp/esp8266/projects/espterm-firmware/esp_iot_sdk_v1.5.2/lib/libnet80211.a /home/myuser/dev/projects/c/esp/esp8266/projects/espterm-firmware/esp_iot_sdk_v1.5.2/lib/libwpa.a /home/myuser/dev/projects/c/esp/esp8266/projects/espterm-firmware/esp_iot_sdk_v1.5.2/lib/liblwip.a /home/myuser/dev/projects/c/esp/esp8266/projects/espterm-firmware/esp_iot_sdk_v1.5.2/lib/libssl.a /home/myuser/dev/projects/c/esp/esp8266/projects/espterm-firmware/esp_iot_sdk_v1.5.2/lib/libupgrade.a /home/myuser/dev/projects/c/esp/esp8266/projects/espterm-firmware/esp_iot_sdk_v1.5.2/lib/libnet80211.a /home/myuser/dev/projects/c/esp/esp8266/projects/espterm-firmware/esp_iot_sdk_v1.5.2/lib/liblwip.a /home/myuser/dev/projects/c/esp/esp8266/projects/espterm-firmware/esp_iot_sdk_v1.5.2/lib/libphy.a /home/myuser/dev/projects/c/esp/esp8266/projects/espterm-firmware/esp_iot_sdk_v1.5.2/lib/libcrypto.a /home/myuser/dev/projects/c/esp/esp8266/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/libgcc.a -T /home/myuser/dev/projects/c/esp/esp8266/projects/espterm-firmware/esp_iot_sdk_v1.5.2/ld/eagle.app.v6.ld -B/home/myuser/dev/projects/c/esp/esp8266/projects/espterm-firmware/esp_iot_sdk_v1.5.2/lib -o image.elf
`

kuzmanzs, 2020-09-07:
issue fixed

closed 2020-09-07 by kuzmanzs

cnlohr, 2020-09-15:
For the sake of those who come after you, what did you have to do to fix it?