cnlohr/channel3 #27: WIFI problem

kuzmanzs, 2020-09-11:
Hi,

I have flashed the device including mpfs. Really am not able to connect to device as AP/STA. stable. sometimes I can connect but the webpage loading is very very slow and I see only Introduction and NTSC button.

IN STA mode I got IP but webside is very slow..

Video is working.

I have tested on ESP8266 Wemos mini and ESP8266-01 as sell, same issue.

[image]

cnlohr, 2020-09-15:
The issue is in probably that your setup, you have a spur which is triggering the CSMA.  I would try other lengths and orientations of the transmitter wire to see if you can find one which mutes the noise bit.

kuzmanzs, 2020-09-16:
I have tried on wemos mini and with ESP8266-01 incl. programmer (https://www.ebay.com/itm/AZDelivery-USB-zu-ESP8266-01-Serial-Wireless-Wifi-Module-fur-ESP-01/143651482795?hash=item21724bacab:g:Sd0AAOSwArNfRjhL)

I have on both devices strong WIFI issues.

Please, could you send me your setup?
Whitch device do you use? Do you power it from programmer or? Did you removed the capacitor between IC output and antenna?

cnlohr, 2020-11-30:
I have had it work on both NodeMCU and ESP-01, both have very bad wifi issues when running on channel 3, and both I needed to do different things with to attenuate the signal sufficiently.