cnlohr/channel3 #29: Error

NikunjChiluvuri, 2020-10-27:
When i type make all, it gives me this error:
common.mf:53: Warning: No shell variable 'ESP_ROOT', using '/home/nikunj/esp8266/esp-open-sdk'
ome/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -mlongcalls -Os -Iome/esp8266/esp_nonos_sdk/include -Iesp82xx/include -I. -Iesp82xx/fwsrc -Iuser -Itablemaker -DICACHE_FLASH -DUSE_OPTIMIZE_PRINTF -DMFS_PAGE_OFFSET=524288 -DSOFTAP_CHANNEL=11 -I. -Iinclude -DWEB_PORT=80 -DCOM_PORT=7777 -DBACKEND_PORT=7878  -DSLOWTICK_MS=50 -DVERSSTR='"Version: 6edcb-dev - Build Tue, Oct 27 2020, 15:48:12 +0530 with -DICACHE_FLASH -DUSE_OPTIMIZE_PRINTF -DMFS_PAGE_OFFSET=524288 -DSOFTAP_CHANNEL=11 -I. -Iinclude -DWEB_PORT=80 -DCOM_PORT=7777 -DBACKEND_PORT=7878  -DSLOWTICK_MS=50"' -g esp82xx/fwsrc/uart.c esp82xx/fwsrc/esp82xxutil.c esp82xx/fwsrc/flash_rewriter.c esp82xx/fwsrc/http.c esp82xx/fwsrc/commonservices.c esp82xx/fwsrc/http_custom.c esp82xx/fwsrc/mdns.c esp82xx/fwsrc/mfs.c user/custom_commands.c user/user_main.c user/ntsc_broadcast.c user/3d.c tablemaker/broadcast_tables.c tablemaker/CbTable.c -flto -Wl,--relax -Wl,--gc-sections -nostdlib -Lome/esp8266/esp_nonos_sdk/lib -Lesp82xx/libgcc_stripped.a ome/esp8266/esp_nonos_sdk/lib/libmain.a ome/esp8266/esp_nonos_sdk/lib/liblwip.a ome/esp8266/esp_nonos_sdk/lib/libssl.a ome/esp8266/esp_nonos_sdk/lib/libupgrade.a ome/esp8266/esp_nonos_sdk/lib/libnet80211.a ome/esp8266/esp_nonos_sdk/lib/libwpa.a ome/esp8266/esp_nonos_sdk/lib/libphy.a ome/esp8266/esp_nonos_sdk/lib/liblwip.a ome/esp8266/esp_nonos_sdk/lib/libcrypto.a ome/esp8266/esp_nonos_sdk/lib/libc.a ome/esp8266/esp_nonos_sdk/lib/libespnow.a ome/esp8266/esp_nonos_sdk/lib/libpp.a esp82xx/libgcc_stripped.a -T ome/esp8266/esp_nonos_sdk/ld/eagle.app.v6.ld -T ome/esp8266/esp_nonos_sdk/ld/eagle.rom.addr.v6.ld -Bome/esp8266/esp_nonos_sdk/lib  -o image.elf
make: ome/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc: Command not found
main.mf:134: recipe for target 'image.elf' failed
make: *** [image.elf] Error 127
I'm using ubuntu 16.4 LTS.

aaronjamt, 2020-11-14:
Well, it's not able to find `ome/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc`. It seems like something is misconfigured and it should be `/home/nikunj/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc` (notice `/home/nikunj/` vs `ome/`). Check your `user.cfg`, I think the important line is `ESP_DEFAULT  = $(HOME)/esp8266/esp-open-sdk`. Has that been changed? Also, check your `HOME` enviroment variable (`echo $HOME` from your shell should do it) and make sure that points to your home directory (looks like it should be `/home/nikunj` based on the log you posted). Does that help?

NikunjChiluvuri, 2020-11-14:
oh, i see. i will try it and replay

NikunjChiluvuri, 2020-11-14:
and yes i have changed it to direct it to my compiller and i think i miscomfigered it.

NikunjChiluvuri, 2020-11-14:
i know what will happen if the directry is different, but i did not see that it is pointing to another directry.

NikunjChiluvuri, 2020-11-14:
ok i tried it and I"m now getting another error : 
esp82xx/common.mf:53: Warning: No shell variable 'ESP_ROOT', using '/home/pi/esp8266/esp-open-sdk'
mkdir -p web/page user; \
cp -n esp82xx/user.cfg.example user.cfg; \
touch user.cfg
mkdir -p web/page user; \
cp -n esp82xx/Makefile.example Makefile; \
touch Makefile
esp82xx/common.mf:53: Warning: No shell variable 'ESP_ROOT', using '/home/pi/esp8266/esp-open-sdk'
cp /home/pi/esp8266/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/libgcc.a esp82xx/libgcc_stripped.a
/home/pi/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-ar dv esp82xx/libgcc_stripped.a _muldi3.o
/home/pi/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-ar: 1: /home/pi/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-ar: @8: not found
/home/pi/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-ar: 1: /home/pi/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-ar:ELF: not found
/home/pi/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-ar: 24: /home/pi/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-ar: p+k[: not found
/home/pi/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-ar: 25: /home/pi/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-ar: x+z[: not found
/home/pi/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-ar: 26: /home/pi/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-ar: ▒+▒Y: not found
/home/pi/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-ar: 27: /home/pi/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-ar: ▒+▒[: not found
/home/pi/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-ar: 28: /home/pi/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-ar: Syntax error: "(" unexpected
make: *** [esp82xx/main.mf:105: esp82xx/libgcc_stripped.a] Error 2

NikunjChiluvuri, 2020-11-14:
and now i dont have a ubantu systen so I'm using my raspberry pi zero w.

cnlohr, 2020-11-14:
I strongly recommend you follow this guide here to setup your environment: https://www.youtube.com/watch?v=LiKNRBIcGII

cnlohr, 2020-11-14:
You're going to need an environment in which you can run those tools.

NikunjChiluvuri, 2020-11-14:
oh hi

NikunjChiluvuri, 2020-11-14:
ok

NikunjChiluvuri, 2020-11-14:
and i followed that guide still I'm getting error

aaronjamt, 2020-11-14:
Can you run `file /home/pi/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-ar` and `uname -a` on the RPi? I suspect the binaries are built for x86_64 but the RPis are arm based. What systems and OS's do you have access to now?

NikunjChiluvuri, 2020-11-15:
i have access to windows 10 only

NikunjChiluvuri, 2020-11-15:
and i dont want to run a vm again.. it is making my system veryyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyy slow

NikunjChiluvuri, 2020-11-15:
and wsl1 isnt running on my windows pc. do you know how to install wsl1?

NikunjChiluvuri, 2020-11-15:
i will try to download it.

aaronjamt, 2020-11-16:
Uhh... you're gonna have to google for WSL1 instructions (or ask @cnlohr) because I don't know what the difference between it and WSL2 is and I don't have much experience with it, sorry.

NikunjChiluvuri, 2020-11-16:
ok

cnlohr, 2020-11-17:
The video I posted above is in Windows 10.  You can install WSL 1 here: https://docs.microsoft.com/en-us/windows/wsl/install-win10

NikunjChiluvuri, 2020-11-17:
ok i will see

NikunjChiluvuri, 2020-11-17:
it got stuck at Installing, this may take a few minutes...  : - |

NikunjChiluvuri, 2020-11-19:
and how to get the c programs?

NikunjChiluvuri, 2020-11-20:
i will close this for now.

closed 2020-11-20 by NikunjChiluvuri