cnlohr/channel3 #30: Missing file

NikunjChiluvuri, 2020-10-27:
When i see the other issues, they had a file called image.elf-0x10000.bin and when i download the code, it it missing. can i get the file?

NikunjChiluvuri, 2020-10-28:
ok.... i got hold of the file. closing

closed 2020-10-28 by NikunjChiluvuri

cnlohr, 2020-11-30:
I didn't realize it wasn't in here.  Can you post instructions, @NikunjChiluvuri

NikunjChiluvuri, 2020-12-01:
oh hi i did not see that you commented as i closed this post. Sorry for the late responce. ok so this is the problume. i saw like 20 issus and they had an extra file. and this is an closed issue. the problume is now solved. i did not test it yet. i'm going to test it on a VM as WSL1 installer is stuck at Installing please wait for more than an hour. I will try installing WSL1 on a another Windows 10 Laptop. so shoud i re open the issue?