cnlohr/channel3 #31: ESP8266 with only 1MB of flash

aaronjamt, 2020-11-14:
I have followed the instructions in the videos, the README.md's, and looked at other issues, but have not been able to get my chip to work. I get no image or sound from the TV and see no new WiFi networks (definitly not ESP****, but have not seen anything else that is open). I think I have found the issue: I am using ESP8266-01 chips with 1MB of flash but the software expects more (I'm guessing 8MB). When running `make burn` or `make burnitall` it flashed fine (at least, it did not crash and the exit code was 0) but would not work (I now see that it auto-detected the chip size as 8MB, so that's probably why it was fine). I realized that the `esptool.py` it was using was an older version, so symbolic linked my system-wide copy to `~/esp8266/esp-open-sdk/esptool/esptool.py`. With this newer version, it correctly auto-detects my 1MB chip and fails, stating that "A fatal error occurred: File {my home directory}/channel3/esp82xx/toolchain/esp_nonos_sdk/bin/blank.bin (length 4096) at offset 4173824 will not fit in 1048576 bytes of flash. Use --flash-size argument, or change flashing address". Is there any way that I can modify this to be compatible or does it simply need more than 1MB of flash to store everything? Thanks in advance.

NikunjChiluvuri, 2020-11-15:
it dose'nt give wifi. and your esp shoud have an external antina connected. and your tv shoud support analog mode. and you can't modify it. it is an binary program and if you were able to read it, it is not in C or C++. it will be a compleatly different language.

cnlohr, 2020-11-19:
Ooph... I really don't recall those ins and outs, I've only ever tested this on >1MB Flash.

cnlohr, 2020-11-19:
(Anyone else who knows can comment) but @NikunjChiluvuri is incorrect.
(1) It should broadcast an AP, but will likely be very difficult to receive because of the analog broadcast.
(2) It is written in C, and can be recompiled from whence using this repository.

aaronjamt, 2020-11-19:
Sorry for such a late reply, I forgot to hit send.

> Ooph... I really don't recall those ins and outs, I've only ever tested this on >1MB Flash.
RIP me, lol. I only have ESP8266-01's that I bought off Amazon in a 5 or 10 pack (don't really remember, long time ago) and 2 NodeMCU ESP32S's, but only one works (I fried the other :cry:), it's in use, and, even if it wasn't, or if the other worked, this only works on ESP8266's (see #20 and #16)

> (1) It should broadcast an AP, but will likely be very difficult to receive because of the analog broadcast.
That's what I guessed (based on the minimal amount of the source I read though not very good with ESP C, only really used Arduino IDE, as well as the README.md for esp82xx)
> (2) It is written in C, and can be recompiled from whence using this repository.
Yep, that's what GitHub says (79.6% C)

NikunjChiluvuri, 2020-11-19:
ok sorry

NikunjChiluvuri, 2020-11-19:
i will check it
and close mine

NikunjChiluvuri, 2020-11-19:
and i cant install wsl1

NikunjChiluvuri, 2020-11-19:
and how to get the c programs?

NikunjChiluvuri, 2020-11-19:
and if you can, give me the link.

NikunjChiluvuri, 2020-11-19:
hello,
    I found this copy of channel3 made in arduino IDE. Here is the link : https://github.com/Alexander12827/chlib

Warning! : I did not test this libray. I'm not sure that this works. I went through the code, and found out that this does'nt use wifi.

aaronjamt, 2020-11-19:
> hello,
> I found this copy of channel3 made in arduino IDE. Here is the link : https://github.com/Alexander12827/chlib
> 
> Warning! : I did not test this libray. I'm not sure that this works. I went through the code, and found out that this does'nt use wifi.

Wow, thanks! I'll try that, it might work. I think the issue it the fact that this uses @cnlohr 's custom SDK which relies on certain parts being in certain places in the flash. Thanks for pointing that other repo out @NikunjChiluvuri !

aaronjamt, 2020-11-19:
I'm going to close this for now, since it seems that it would need major rework to work on a 1MB chip, and since @NikunjChiluvuri pointed out an alternative.

closed 2020-11-19 by aaronjamt

NikunjChiluvuri, 2020-11-20:
Your welcome :)

NikunjChiluvuri, 2020-11-20:
> (Anyone else who knows can comment) but @NikunjChiluvuri is incorrect.
> (1) It should broadcast an AP, but will likely be very difficult to receive because of the analog broadcast.
> (2) It is written in C, and can be recompiled from whence using this repository.

wait what?! it gives me an wifi connection?!

NikunjChiluvuri, 2020-11-20:
ok nvm i now understand