cnlohr/channel3 #38: .bin dump instead of installing it through ESP82xx?

CowMowTV, 2022-08-24:
I once again tried to install ESP82xx only to find out that WSL isn‘t working properly on Windows 11 and that installing ESP82xx with Linux also doesn‘t work with my ESP. 
My old ESP got, well, toasted (was a 3$ Chinese one), and so I decided to try the project out with better equipment like a soldering iron and basic band pass. Because I had a spare one laying around, I tried installing Channel3.

Wouldn‘t it just be easier to provide a dumped version of Channel3 instead of having to install dozens of things just to get it running, possibly having corrupted something in the process?

cnlohr, 2022-11-09:
I haven't worked on this in a very long time, but if I do I can include a binary.

AyceBenz, 2024-06-28:
Oh you need to.