cnlohr/channel3 #61: ESP8266 Spewing out errors to serial

byronbonkers, 2023-07-31:
I need some help flashing, my esp just spews out errors in the serial monitor after flashing channel3, it doesn't do anything either, I have tried wiping the flash and trying different flashers, but still no, it will work with a normal arduino sketch though.

Its saying 
 ```
ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 28168, room 16 
tail 8
chksum 0xf3
load 0x3ffe8000, len 5240, room 0 
tail 8
chksum 0x83
load 0x3ffe9480, len 5212, room 0 
tail 12
chksum 0x28
csum 0x28
mismatch map 2,spi_size_map 4
boot not set
ota1 not set
ota2 not set
system_partition_table_regist 4 success!!
>> 0x3fc,0x3fc000rf_cal[0] !=0x05,is 0xFF

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 28168, room 16 
tail 8
chksum 0xf3
load 0x3ffe8000, len 5240, room 0 
tail 8
chksum 0x83
load 0x3ffe9480, len 5212, room 0 
tail 12
chksum 0x28
csum 0x28
mismatch map 2,spi_size_map 4
boot not set
ota1 not set
ota2 not set
system_partition_table_regist 4 success!!
>> 0x3fc,0x3fc000rf_cal[0] !=0x05,is 0xFF

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 28168, room 16 
tail 8
chksum 0xf3
load 0x3ffe8000, len 5240, room 0 
tail 8
chksum 0x83
load 0x3ffe9480, len 5212, room 0 
tail 12
chksum 0x28
csum 0x28
mismatch map 2,spi_size_map 4
boot not set
ota1 not set
ota2 not set
system_partition_table_regist 4 success!!
>> 0x3fc,0x3fc000rf_cal[0] !=0x05,is 0xFF

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 28168, room 16 
tail 8
chksum 0xf3
load 0x3ffe8000, len 5240, room 0 
tail 8
chksum 0x83
load 0x3ffe9480, len 5212, room 0 
tail 12
chksum 0x28
csum 0x28
mismatch map 2,spi_size_map 4
boot not set
ota1 not set
ota2 not set
system_partition_table_regist 4 success!!
>> 0x3fc,0x3fc000rf_cal[0] !=0x05,is 0xFF

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 28168, room 16 
tail 8
chksum 0xf3
load 0x3ffe8000, len 5240, room 0 
tail 8
chksum 0x83
load 0x3ffe9480, len 5212, room 0 
tail 12
chksum 0x28
csum 0x28
mismatch map 2,spi_size_map 4
boot not set
ota1 not set
ota2 not set
system_partition_table_regist 4 success!!
>> 0x3fc,0x3fc000rf_cal[0] !=0x05,is 0xFF

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 28168, room 16 
tail 8
``` and the blue light just flashing repeatedly

cnlohr, 2023-07-31:
Try `make erase burnitall` At least I think it's erase.

byronbonkers, 2023-07-31:
but I have re-downloaded and tried over with redownloading the git repo then starting the commands again, won't this just do the same thing still?

byronbonkers, 2023-07-31:
OMFG I think it worked now, will it also have the web interface?

byronbonkers, 2023-07-31:
uhh do u have discord so its easier to msg u, but also, the esp's wifi hotspot pops up but when i connect and visit 192.168.4.1 it just doesn't work

cnlohr, 2023-08-01:
Do you get a valid IP?  What does come up?  And please join the discord, others can probably answer some of these questions faster than me.