cnlohr/channel3 #64: How to Build and flash?

AyceBenz, 2024-06-28:
For me and anybody new to this,how do I use these files? What software and Is and how to create the FW image/ flash it into a board

0wwafa, 2024-12-17:
Yep. if possible provide a BIN file to flash!

MichaelK-F, 2024-12-18:
I tried to build the bin files, but I don't have an esp to test with at the moment. Tell me if it works.

[bin-files.zip](https://github.com/user-attachments/files/18174801/bin-files.zip)

AyceBenz, 2024-12-18:
thank you, i will check the results when i finish when i have time.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

AyceBenz, 2024-12-18:
stupid avast

AyceBenz, 2024-12-23:
> I tried to build the bin files, but I don't have an esp to test with at the moment. Tell me if it works.
> 
> [bin-files.zip](https://github.com/user-attachments/files/18174801/bin-files.zip)

didnt seem to do anything
cant even control it,no AP,no anything
still havnt tried it with a tv as ours dont have antenna yet